uzh-rpg/rpg_svo_example #1: Download link is empty

electricmanlpl, 2017-07-18:
Hi,
Thanks for releasing the exciting academic work, I am glad to test it, but it seems the binary link is empty.

zhangzichao, 2017-07-18:
Hi, did you fill in the form following the link (http://rpg.ifi.uzh.ch/svo2.html)? You should receive an email containing the download links after that.

electricmanlpl, 2017-07-19:
@zhangzichao when I click the link, it's like this
[image]
There's no any form...

zhangzichao, 2017-07-19:
Just click the link under the second paragraph... sorry maybe we should make it more explicit

electricmanlpl, 2017-07-19:
I did click the link, it's straightforward... I mean after I click it and enter a new page, there's nothing except the contents:
[image]

zhangzichao, 2017-07-19:
Ah, it should be something like this
[image]
Maybe trying another browser? Or wait a bit for it to load if the internet is slow.

closed 2017-08-18 by zhangzichao