uzh-rpg/rpg_svo_example #2: catkin build error

longchao343, 2017-07-20:
Hi,
      I was building the binary on ubuntu16.04, but errors happend.
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target svo_msgs_generate_messages_cpp
[  0%] Built target svo_msgs_generate_messages_nodejs
Scanning dependencies of target svo_msgs_generate_messages_py
Scanning dependencies of target dynamic_reconfigure_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_gencfg
Scanning dependencies of target pcl_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target svo_msgs_generate_messages_py
[  0%] Built target pcl_msgs_generate_messages_cpp
Scanning dependencies of target bond_generate_messages_nodejs
Scanning dependencies of target pcl_msgs_generate_messages_lisp
Scanning dependencies of target visualization_msgs_generate_messages_eus
Scanning dependencies of target dynamic_reconfigure_generate_messages_eus
[  0%] Built target pcl_msgs_generate_messages_lisp
[  0%] Built target bond_generate_messages_nodejs
[  0%] Built target visualization_msgs_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_eus
Scanning dependencies of target pcl_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_lisp
Scanning dependencies of target topic_tools_generate_messages_lisp
Scanning dependencies of target topic_tools_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target pcl_msgs_generate_messages_py
[  0%] Built target topic_tools_generate_messages_lisp
[  0%] Built target topic_tools_generate_messages_nodejs
Scanning dependencies of target svo_ros
[ 11%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_interface.cpp.o
[ 22%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_node_base.cpp.o
[ 33%] Building CXX object CMakeFiles/svo_ros.dir/src/visualizer.cpp.o
[ 44%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_factory.cpp.o
[ 55%] Linking CXX shared library /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/devel/lib/libsvo_ros.so
[ 55%] Built target svo_ros
Scanning dependencies of target svo_nodelet
Scanning dependencies of target svo_node
[ 66%] Building CXX object CMakeFiles/svo_nodelet.dir/src/svo_nodelet.cpp.o
[ 77%] Building CXX object CMakeFiles/svo_node.dir/src/svo_node.cpp.o
[ 88%] Linking CXX executable /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/devel/lib/svo_ros/svo_node
[100%] Linking CXX shared library /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/devel/lib/libsvo_nodelet.so
[100%] Built target svo_nodelet
/usr/bin/ld: warning: libopencv_core3.so.3.2, needed by /home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_highgui3.so.3.2, needed by /home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_imgproc3.so.3.2, needed by /home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_imgcodecs3.so.3.2, needed by /home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libvikit_cameras.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_calib3d3.so.3.2, needed by /home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libvikit_common.so, not found (try using -rpath or -rpath-link)
/home/x/workspace/svo_binaries_1604_kinetic/svo_install_ws/install/lib/libvikit_common.socv::findHomography(cv::_InputArray const&, cv::_InputArray const&, int, double, cv::_OutputArray const&, int, double)
[svo_ros] <== '/home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                                                                                                                                                        
Failed   <== svo_ros [ 37.0 seconds ]                                                                                                                                                                                                  
[build] There were '1' errors:                                                                                                                                                                                                         

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/build/svo_ros && /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/x/workspace/svo_binaries_1604_kinetic/svo_install_overlay_ws/build/build_logs/svo_ros.log

Exited with return code: 2 
                                                                                                                                                                                                                                       
[build] Runtime: 37.2 seconds

zhangzichao, 2017-07-20:
Hi,

It seems it cannot find the opencv3 libraries shipped with ROS.

Can you post the configuration of  your `svo_install_overlay_ws` by typing `catkin config` ?

Also check if the opencv3 libraries are there: `ls -l /opt/ros/kinetic/lib/ | grep opencv`.

longchao343, 2017-07-21:
Hi,
       I think I need to build opencv3.2 on my computer, because opencv libaries version on my computer is 3.2 :(

closed 2017-08-18 by zhangzichao