uzh-rpg/rpg_svo_example #4: Scaramuzza's camera model?

mark-eth, 2017-07-22:
Why don't support Scaramuzza's camera model?

https://sites.google.com/site/scarabotix/ocamcalib-toolbox

I think it's a very high accurate camera model for all cameras (small FOV, fisheye, and omnidirectional cameras)

zhangzichao, 2017-07-24:
Hi, 

we do support the camera model you mentioned, please refer to the last section of the document
https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/doc/camera_calibration.md

xuqingwenkk, 2017-09-07:
I tried to use `omni_matlab_to_rpg.py` to convert the matlab calibration result. However, the number of my intrinsics data is 27, more than 24, because `cam_params['invpol']` has 15 parameters. Then I modify intrinsics cols to 27, in this case the process will died as follows.
`[ INFO] [1504773960.834672909]: Found parameter: calib_file, value: /home/xuqw/svo/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/calib/bluefox_20170907_fisheye.yaml
E0907 16:46:00.834779  2918 ncamera.cpp:54] Failed to load NCamera from file /home/xuqw/svo/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/calib/bluefox_20170907_fisheye.yaml with the error: 
yaml-cpp: error at line 0, column 0: bad file
*** Aborted at 1504773960 (unix time) try "date -d @1504773960" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGSEGV (@0x20) received by PID 2918 (TID 0x7f7cd795c7c0) from PID 32; stack trace: ***
    @     0x7f7cd67324b0 (unknown)
    @     0x7f7cd7195a12 std::vector<>::size()
    @     0x7f7cd71c8302 vk::cameras::NCamera::numCameras()
    @     0x7f7cd71c6a67 svo::factory::loadCameraFromYaml()
    @     0x7f7cd71c78ab svo::factory::makeMono()
    @     0x7f7cd718b272 svo::SvoInterface::SvoInterface()
    @     0x7f7cd71cfb33 svo_ros::SvoNodeBase::SvoNodeBase()
    @           0x4011c4 main
    @     0x7f7cd671d830 __libc_start_main
    @           0x4010a9 _start
    @                0x0 (unknown)
[svo-2] process has died [pid 2918, exit code -11, cmd /home/xuqw/svo/svo_install_overlay_ws/devel/lib/svo_ros/svo_node __name:=svo __log:=/home/xuqw/.ros/log/f7794bf0-93a8-11e7-9c44-64006a6556ea/svo-2.log].
log file: /home/xuqw/.ros/log/f7794bf0-93a8-11e7-9c44-64006a6556ea/svo-2*.log`
What should I do to avoid this error?

Razbotics, 2018-03-27:
same issue , did you solve the problem?

xuqingwenkk, 2018-03-28:
Yes. Limit the inverse polynomial order in file 'findinvpoly,m' of the calibration toolbox, then you can solve it.

xuqingwenkk, 2018-03-28:
Yes. Limit the inverse polynomial order in file 'findinvpoly,m' of the calibration toolbox, then you can solve it. Thanks for Zhang Zichao's help.

gftabor, 2018-03-28:
I found better results using the new matlab fisheye checkerboard detector. 2018a adds a tool to use it and 2017b had full functionality just without tool.

Then I exported the model using the same steps described by the SVO2 documentation.

ustccxy, 2024-11-22:
> 是的。限制校准工具箱的 'findinvpoly，m' 文件中的多项式反阶数，然后就可以求解它了。感谢张子超的帮助。

hello，可以详细说一下解决方法吗？