uzh-rpg/rpg_svo_example #5: In EuRoC V101 dataset, the estimated trajectory has a big error!

highlightz, 2017-07-24:
Hi, I use the default pinhole.yaml config file and test V101 dataset with my machine Intel® Core™ i5-4570 CPU @ 3.20GHz × 4 . Sometimes the estimator runs through to the end without tracking failed, sometimes, it cannot run to the end. For the first situation, I've got the below trajectory:
[image]
And for the latter situation, I got the below trajectory, with a big error:
[image]
scale: 2.650539 
compared_pose_pairs 1651 pairs
absolute_translational_error.rmse 0.539003 m
absolute_translational_error.mean 0.464468 m
absolute_translational_error.median 0.451905 m
absolute_translational_error.std 0.273484 m
absolute_translational_error.min 0.101481 m
absolute_translational_error.max 1.536845 m
How can I improve the performance by tuning the parameters?
As a matter of fact, I notice in your paper that, your SVO2 program runs well on all EuRoC datasets. Shown as below chart:
[image]

zhangzichao, 2017-07-24:
Hi, you will need to change the parameters. I will add some example parameter files for EUROC as well as a general instruction for parameters soon.

highlightz, 2017-07-25:
Okay, Thanks.

zhangzichao, 2017-08-18:
Hi, we added some example launch files and parameters for EuRoC datasets. Should be better than the default one and also helps with the issue #3 .

highlightz, 2017-08-24:
Got it and thank you. I've tried them with EuRoC datasets, it is much better than before, though there is still big drift at V1 and V2 datasets.

JohanVer, 2017-08-24:
Hi, I got everything to run and just tried the EuRoC dataset MH_01_easy in stereo mode using the launch file: euroc_stereo_imu.launch .
Thereby I got some questions:
1.Though it tracks the position of the camera I get lots of:  "seed sigma is nan!-0.338734, sq-nan, check-convergence = 0"  and "critical". What does that mean? 
2. I also wondered if the baseline of the stereo isn't provided somewhere (does it estimate scale in stereo mode)?
3. Is there there any need to pre-undistort or rectify the input images or is it done by the tool?

Thanks a lot for the great software!

zhangzichao, 2017-08-25:
Hi, 
1. it means something wrong with the depth filter, we know this issue and hope to fix/suppress it in the future.
2. the baseline is defined implicitly in the [calibration file](https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/calib/euroc_stereo_calib.yaml). The `T_B_C` defines the transformation from the body frame to the camera frame ([transformation convention](https://github.com/uzh-rpg/rpg_svo/wiki/Notation)).
3. No, there is no need for undistortion. The camera distortion is already taken care of internally.

JohanVer, 2017-08-25:
Hey, great thank you very much for your fast answer!
1. Can I for now ignore the error or is it known that this problem influences the tracking accuracy significantly?
2. Regarding the baseline: 
I have kalibr calibration files (yaml) and converted them to the svo format using your provided script. The kalibr calibration includes the matrix T_cn_cnm1 which should give the transformation between the right camera and the left camera. The converter sets both T_B_C matrices to identity because of "No IMU is specified, so T_B_C is set to identity". So I should overwrite the T_B_C of the right camera in the svo-yaml with T_cn_cnm1 and leave the left T_B_C with the identity matrix? Is this correct?