uzh-rpg/rpg_svo_example #7: Position calculation problem

hosh0425, 2017-08-12:
Hi . I am using a monocular camera. I move the camera in each three axes about 2meter but the change in topic /pose/camera is very less than 2meter(about 0.4). Should I scale this data by my self?

zhangzichao, 2017-08-18:
Yes, for monocular case you do not have the exact scale anyway.  The parameter `map_scale` defines the median  of the scene depth for initialization and the scale is carried on afterwards.

What you can do is to initialize the pipeline by pointing your camera perpendicular to a plane and moving the camera in parallel to it. Then if you know how far the plane is from the camera, say 1 meter, you can set the `map_scale` to 1. This should gives you a roughly correct scale.

bharat-robotics, 2018-02-04:
I am also having a similar issue while moving clearpath husky around the corridor taking rounds (using monocular camera). **However, the scale decreases when I move the robot around.** Are there any specific parameters I should look at? I have attached some pictures for reference.
[image]
[image]