uzh-rpg/rpg_svo_example #8: Additional architecture support

gftabor, 2017-08-23:
Are there plans to release binaries for more CPU architectures? Or giving access to the source code so people could compile their own? Specifically I had been hoping to try to get binaries for the Nvidia Jetson TX2. 
https://devblogs.nvidia.com/parallelforall/jetson-tx2-delivers-twice-intelligence-edge/ 

If you already cross compiled for the Odroid XU4 you probably have most of the work out of the way to compile for Armv8. The binaries from the XU4 would probably be helpful to some people as well.

zhangzichao, 2017-08-24:
Hi, unfortunately we do not have binaries ready for arm architecture now. We plan to prepare it in late September.

hwiryong-jung, 2017-09-29:
I'm also waiting for ARM architecture support : )

zhangzichao, 2017-09-29:
Hi, we just put the binaries for armhf (ubuntu 14.04 32bit with ROS indigo) online, you can check the [readme](https://github.com/uzh-rpg/rpg_svo_example/blob/master/README.md) for details, hope it works for you.

@gftabor unfortunately we do not have crosscompiling setup, the binaries are built on Odroid XU4 natively (also tested on Odroid XU3). Therefore it probably does not work for you if you have a 64bit system on armv8. If you know some good instructions for crosscompiling for different arm architecures/OS, I would be happy to try, which would also makes future release easier. Thanks.

hwiryong-jung, 2017-09-30:
Wow, thank you!! The binaries for armhf (ubuntu 16.04 64bit with ROS kinetic) also available? : )
In my case, running ubuntu 14.04 binaries had problem, so I'm currently using ubuntu 16.04 version.

sulim-vladimir, 2017-11-09:
Thank you for sharing your significant work. Will there be a version for 32-bit Ubuntu 16.04? I tried SVO2 on x86_64 and it runs perfectly, but on i686 it can't be built. Thank you.

MrDadaGuy, 2018-08-21:
Also trying to get this working on Nvidia Jetson TX2 / 16.04 Kinetic... A release for this platform would be great.  Thanks!