uzh-rpg/rpg_svo_example #10: exit code -11 ????

hwiryong-jung, 2017-08-28:
During following "install.pdf", 

roslaunch svo_ros run_from_topic.launch cam_name:=svo_test_pinhole
rosbag play svo_test_short.bag

Executing first line of above commands shows me messages below ..

NODES
  /
    svo (svo_ros/svo_node)
    svo_gui (rqt_gui/rqt_gui)
    vis (rviz/rviz)

auto-starting new master
process[master]: started with pid [6505]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9f9e9f04-8bc2-11e7-ab49-10bf4802e9ef
process[rosout-1]: started with pid [6518]
started core service [/rosout]
process[svo-2]: started with pid [6536]
process[vis-3]: started with pid [6537]
process[svo_gui-4]: started with pid [6538]
[ INFO] [1503905368.353618231]: Found parameter: pipeline_is_stereo, value: 0
[ WARN] [1503905368.354739359]: Cannot find value for parameter: set_initial_attitude_from_gravity, assigning default: 1
[ WARN] [1503905368.355608147]: Cannot find value for parameter: automatic_reinitialization, assigning default: 0
[ INFO] [1503905368.358716792]: Found parameter: calib_file, value: /home/xxxx/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/calib/svo_test_pinhole.yaml
loaded 1 cameras
  name = cam
  size = [752, 480]
  Projection = Pinhole
  Focal length = (383.013, 382.392)
  Principal point = (344.706, 244.327)
  Distortion: Atan(0.932)
[ WARN] [1503905368.360813677]: Cannot find value for parameter: klt_max_level, assigning default: 4
[ WARN] [1503905368.361654246]: Cannot find value for parameter: klt_min_level, assigning default: 0
[ INFO] [1503905368.364081464]: Found parameter: reprojector_max_n_kfs, value: 5
[ INFO] [1503905368.367630233]: Found parameter: max_fts, value: 180
[ INFO] [1503905368.369590566]: Found parameter: grid_size, value: 30
[ WARN] [1503905368.370461410]: Cannot find value for parameter: reproject_unconverged_seeds, assigning default: 1
[ INFO] [1503905368.372960380]: Found parameter: reprojector_affine_est_offset, value: 1
[ INFO] [1503905368.376066977]: Found parameter: reprojector_affine_est_gain, value: 0
[ WARN] [1503905368.377131082]: Cannot find value for parameter: init_min_features, assigning default: 100
[ WARN] [1503905368.378335327]: Cannot find value for parameter: init_min_tracked, assigning default: 80
[ WARN] [1503905368.379482218]: Cannot find value for parameter: init_min_inliers, assigning default: 70
[ INFO] [1503905368.382400624]: Found parameter: init_min_disparity, value: 30
[ WARN] [1503905368.383003656]: Cannot find value for parameter: init_min_features_factor, assigning default: 2
[ WARN] [1503905368.383619175]: Cannot find value for parameter: reproj_err_thresh, assigning default: 2
[ WARN] [1503905368.384210617]: Cannot find value for parameter: init_disparity_pivot_ratio, assigning default: 0.5
[ WARN] [1503905368.384806852]: Cannot find value for parameter: init_method, assigning default: FivePoint
[ INFO] [1503905368.386782697]: Found parameter: grid_size, value: 30
[ INFO] [1503905368.388588079]: Found parameter: n_pyr_levels, value: 3
[ INFO] [1503905368.390797398]: Found parameter: detector_threshold_primary, value: 10
[ INFO] [1503905368.395400837]: Found parameter: detector_threshold_secondary, value: 200
[ WARN] [1503905368.396639542]: Cannot find value for parameter: use_edgelets, assigning default: 1
[ INFO] [1503905368.400412513]: Found parameter: n_pyr_levels, value: 3
[ WARN] [1503905368.401830056]: Cannot find value for parameter: use_threaded_depthfilter, assigning default: 1
[ INFO] [1503905368.405912557]: Found parameter: seed_convergence_sigma2_thresh, value: 200
[ INFO] [1503905368.409179077]: Found parameter: scan_epi_unit_sphere, value: 0
[ INFO] [1503905368.412417123]: Found parameter: depth_filter_affine_est_offset, value: 1
[ INFO] [1503905368.415656376]: Found parameter: depth_filter_affine_est_gain, value: 0
[ INFO] [1503905368.420464838]: Found parameter: max_fts, value: 180
[ WARN] [1503905368.422437965]: Cannot find value for parameter: max_seeds_ratio, assigning default: 3
[ INFO] [1503905368.428156797]: Found parameter: max_n_kfs, value: 30
[ INFO] [1503905368.433160499]: Found parameter: use_imu, value: 0
[ WARN] [1503905368.459941128]: Cannot find value for parameter: trace_dir, assigning default: /home/xxxx/svo_install_ws/install/share/svo/trace
[ WARN] [1503905368.460944007]: Cannot find value for parameter: quality_min_fts, assigning default: 50
[ WARN] [1503905368.461934634]: Cannot find value for parameter: quality_max_drop_fts, assigning default: 40
[ WARN] [1503905368.463312469]: Cannot find value for parameter: relocalization_max_trials, assigning default: 50
[ INFO] [1503905368.467903955]: Found parameter: poseoptim_prior_lambda, value: 0
[ INFO] [1503905368.471746492]: Found parameter: poseoptim_using_unit_sphere, value: 0
[ INFO] [1503905368.473845025]: Found parameter: img_align_prior_lambda_rot, value: 0
[ INFO] [1503905368.475951430]: Found parameter: img_align_prior_lambda_trans, value: 0
[ WARN] [1503905368.476709740]: Cannot find value for parameter: structure_optimization_max_pts, assigning default: 20
[ INFO] [1503905368.478860157]: Found parameter: map_scale, value: 1
[ INFO] [1503905368.481077020]: Found parameter: kfselect_criterion, value: FORWARD
[ WARN] [1503905368.481792754]: Cannot find value for parameter: kfselect_min_dist, assigning default: 0.12
[ INFO] [1503905368.483744476]: Found parameter: kfselect_numkfs_upper_thresh, value: 120
[ INFO] [1503905368.485722160]: Found parameter: kfselect_numkfs_lower_thresh, value: 70
[ INFO] [1503905368.487646365]: Found parameter: kfselect_min_dist_metric, value: 0.1
[ INFO] [1503905368.489510883]: Found parameter: kfselect_min_angle, value: 20
[ INFO] [1503905368.491390025]: Found parameter: kfselect_min_disparity, value: 40
[ WARN] [1503905368.492051439]: Cannot find value for parameter: kfselect_min_num_frames_between_kfs, assigning default: 2
[ INFO] [1503905368.494051356]: Found parameter: img_align_max_level, value: 4
[ INFO] [1503905368.496112193]: Found parameter: img_align_min_level, value: 2
[ WARN] [1503905368.496877473]: Cannot find value for parameter: img_align_robustification, assigning default: 0
[ WARN] [1503905368.497586983]: Cannot find value for parameter: img_align_use_distortion_jacobian, assigning default: 0
[ INFO] [1503905368.501141267]: Found parameter: img_align_est_illumination_gain, value: 0
[ INFO] [1503905368.505531879]: Found parameter: img_align_est_illumination_offset, value: 0
[ INFO] [1503905368.509430554]: Found parameter: poseoptim_thresh, value: 2
[ INFO] [1503905368.513366042]: Found parameter: update_seeds_with_old_keyframes, value: 1
[ INFO] [1503905368.516629768]: Found parameter: use_async_reprojectors, value: 0
[ INFO] [1503905368.516951734]: DepthFilter: created.
[ INFO] [1503905368.517032604]: DepthFilter: Start thread.
[ WARN] [1503905368.518419438]: Cannot find value for parameter: T_world_imuinit/tz, assigning default: 0
[ WARN] [1503905368.519527076]: Cannot find value for parameter: T_world_imuinit/ty, assigning default: 0
[ WARN] [1503905368.520631911]: Cannot find value for parameter: T_world_imuinit/tx, assigning default: 0
[ INFO] [1503905368.530467452]: Found parameter: T_world_imuinit/qz, value: 0
[ INFO] [1503905368.534482020]: Found parameter: T_world_imuinit/qy, value: 0
[ INFO] [1503905368.537775845]: Found parameter: T_world_imuinit/qx, value: 0
[ INFO] [1503905368.541727290]: Found parameter: T_world_imuinit/qw, value: 1
*** Aborted at 1503905368 (unix time) try "date -d @1503905368" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGSEGV (@0x2a) received by PID 6536 (TID 0x7ffb8bae3a40) from PID 42; stack trace: ***
    @     0x7ffb8aaa6cb0 (unknown)
    @     0x7ffb8b1153cb std::string::string()
    @     0x7ffb8b3b5102 svo::Visualizer::Visualizer()
    @     0x7ffb8b39278b svo::SvoInterface::SvoInterface()
    @     0x7ffb8b3b245f svo_ros::SvoNodeBase::SvoNodeBase()
    @           0x400c05 main
    @     0x7ffb8aa91f45 __libc_start_main
    @           0x400c7c (unknown)
    @                0x0 (unknown)
[svo-2] process has died [pid 6536, exit code -11, cmd /home/xxxx/svo_install_overlay_ws/devel/lib/svo_ros/svo_node __name:=svo __log:=/home/xxxx/.ros/log/9f9e9f04-8bc2-11e7-ab49-10bf4802e9ef/svo-2.log].

Can I ask the reason of exit code -11?? : )

QingbiaoLi, 2017-10-22:
Same issue to me?
DId you solve it?

supersheepcy, 2017-11-04:
Me too...how to solve this bug,please?

hwiryong-jung, 2017-11-04:
In my case, I could not resolve this. 
When I traced the root of error, it was from the svo library.
However I think some environmental things could cause error, such as my ROS configuration..
Instead, I set up new PC with ubuntu 16.04, and it works fine now : )

mhubii, 2018-09-05:
On Ubuntu 14.04 you may want to try and install Eigen 3.2.10