uzh-rpg/rpg_svo_example #12: Issues when build svo example

kevin-dong-ros, 2017-09-28:
Hi, I have downloaded the binaries of svo2.0 and installed in ubuntu16.04 followed the support file install.pdf. But when using catkin build to build svo_example after installed svo2.0 binaries, errors happened as flowing picture.
[image]
Who can I solve this error?

lorenwel, 2017-10-06:
I had this issue myself and managed to solve it. 
The problem is that the files in question have their version number appended, which seems to confuse the compiler. If you create symbolic links to those files without the version number everything should compile. 

You can do this by going to `~/svo_install_ws/install/lib/` and executing
```bash
ln -s libgflags.so.2.2.0 libgflags.so
ln -s libglog.so.0 libglog.so
```

kevin-dong-ros, 2017-10-09:
Hi, @lorenwel ,  I have fixed this problem with your method, thanks!

closed 2017-10-09 by kevin-dong-ros

sbyebss, 2019-07-02:
@lorenwel  Hi, Could you please be kind to tell me why svo 2.0 binaries couldn't be downloaded in the robotics& perception lab website? Do you have the svo 2.0 binaries?