uzh-rpg/rpg_svo_example #13: KITTI benchmark

Nickhodem, 2017-09-29:
Hi,
I'm trying to use SVO stereo on KITTI dataset. I changed launch, calib and params files but I don't know how to write T_B_C. Which units should I use? 
It's KITTI hardware setup: http://www.cvlibs.net/datasets/kitti/images/setup_top_view.png
Can you guys can help me?
During test I always get this message:
[ INFO] [1506685381.758927285]: Stereo: Triangulated 48 features,127 failed.
[ERROR] [1506685381.759024257]: Initialization failed. Not enough triangulated points.
What does it mean?
Thanks for your help!
Best Regards
Nickhodem

shmm91, 2018-07-20:
Have you solved this problem? I meet this same issue, could you help me?

Weilaizaijian, 2022-12-05:
> Have you solved this problem? I meet this same issue, could you help me?
Have you solved this problem? I meet this same issue, could you help me?