uzh-rpg/rpg_svo_example #14: REMODE does not work with the message type of the SVO 2.0

Asylbeck, 2017-10-03:
The message type built previously for SVO 1.0 for "/dense_input" does not match the message type of SVO 2.0 and as a result the REMODE does not work with it.

ManashRaja, 2018-06-06:
@Asylbeck  This is because SVO 1.0 publishes "/dense_input" topic of type "svo_msgs/DenseInput" whereas SVO 2.0 publishes that of type "svo_msgs/DenseInputWithFeatures". To solve this:

Edit the following lines in the file `svo_ros/src/visualizer.cpp` to match as written below
75: pub_dense_.at(i) = pnh_.advertise<svo_msgs::DenseInput>("dense_input/"+std::to_string(i), 2);
648: svo_msgs::DenseInput msg;
690: //msg.features.push_back(feature);   //Comment this line out.

Then build the repository.

Haotian-Zhang, 2018-11-02:
@Asylbeck @mpdmanash Hi, have you successfully run the svo 2.0 with REMODE, according to what @mpdmanash said, I've modified the code but it seems the msg dense_input can still not be published. Anything else should be modified? Thank you!