uzh-rpg/rpg_svo_example #16: Camera - IMU coordinate frames

mrsp, 2017-11-15:
Hello 

Could you please clarify the coordinate frames used for the camera/imu?

Is it both NED or do you follow this convention:
https://github.com/ethz-asl/rovio/wiki/Coordinate-Frames-and-Notation

thanks in advance

zhangzichao, 2017-11-22:
With IMU we only determine the initial roll and pitch at the beginning (assume the sensor is static). The yaw angle has nothing to do with the real north direction. The initial frame's position is set to zero.

Then everything afterwards is relative motion with respect to the initial frame. The coordinate frame notation is the same as the open source SVO https://github.com/uzh-rpg/rpg_svo/wiki/Notation

mrsp, 2017-11-22:
Thanks for the reply!

Can you also comment on the max_imu_delta_t param of the imu params? Why is it needed and for what does it stand for? Assuming you have an imu running constantly at 400hz is it 0.5/400 s?

thanks again

zhangzichao, 2017-11-27:
This parameter controls how much tolerance there is for the timestamp difference of the image and corresponding IMU (for example, between the timestamp of the image and the timestamp of the closest IMU measurement).

sven-hoek, 2020-01-09:
What exactly does the parameter `delay_imu_cam` do?