uzh-rpg/rpg_svo_example #18: Problem when running SVO2.0 with external stereo camera

stanleywang-0-1, 2017-11-22:
I'm testing SVO2.0 with an external stereo camera with IMU disabled. Now I have some problem when running SVO via fla_stereo_imu.launch. The program seemed not running properly as there was no trajectory of the camera displayed in Rviz and the information in the shell paused at the lines as shown below(**no new information when I opened my camera**):
[image]
while the program worked when I use the fla_stereo_imu.bag provided(even when I set use_imu to False, it still ran successfully). The information which corresponded to the one paused is as shown below:
[image]

To make use of the fla_stereo_imu.launch, I have made some modifications to match the external stereo camera:
1) I have changed the intrinsic&extrinsic parameters, and set the coordinate frame of the left camera as the body frame.
2) Without using IMU ,I have set the use_imu to false and set the prior lambdas to zero.
3) In fla_stereo_imu.launch, I have changed the topics of the two camera.
Is there anything I forget to do in order to make it work? Thanks a lot.

stanleywang-0-1, 2017-11-22:
I use Ubuntu 16.04 and ROS kinetic.

zhangzichao, 2017-11-22:
Is your stereo camera synchronized? We assume the stereo images are synchronized, as you can see here:
https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/src/svo_interface.cpp#L348

If this is the case, I am not sure you will get good results from unsynchronized images. You can still try by changing the synchronization policy (http://wiki.ros.org/message_filters  see Policy-Based Synchronizer).

stanleywang-0-1, 2017-11-22:
I think the stereo camera is synchronized.
Besides, there was no evidence that the program is running, including the information produced in the shell or the moving camera in Rviz, even it is wrong. I don't think that the synchronization problem would lead to that.

zhangzichao, 2017-11-22:
Difficult to tell just from looking at the terminal output ;)
You can do a quick check by add some log information in this function: https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/src/svo_interface.cpp#L221 to see whether it actually sends the images to svo. Recording a bag to check the timestamps also will help.

stanleywang-0-1, 2017-11-28:
Well, it seems that the image topics published by the camera didn't send images to the topics of svo that receive images. I'm still trying to figure it out.

closed 2017-11-30 by stanleywang-0-1

wowdeath, 2018-03-09:
Have you solved this problem successful? I have met the same question.