uzh-rpg/rpg_svo_example #21: ARM build issue

nightpool, 2017-12-03:
Hi, I'm trying to build on ARM and keep running into the following errors:

```cmake
[ 66%] Building CXX object CMakeFiles/svo_node.dir/src/svo_node.cpp.o
[ 77%] Building CXX object CMakeFiles/svo_nodelet.dir/src/svo_nodelet.cpp.o
[ 88%] Linking CXX executable /home/odroid/svo_install_overlay_ws/devel/lib/svo_ros/svo_node
[100%] Linking CXX shared library /home/odroid/svo_install_overlay_ws/devel/lib/libsvo_nodelet.so
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::output_helper::publishTfTransform(kindr::minimal::QuatTransformationTemplate<double> const&, ros::Time const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tf::TransformBroadcaster&)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `svo::ImuHandler::loadInitializationFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::output_helper::publishCameraMarker(ros::Publisher, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, int, int, double, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `cv::imshow(std::string const&, cv::_InputArray const&)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `YAML::ostream_wrapper::write(std::string const&)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::output_helper::publishPointMarker(ros::Publisher, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, int, int, double, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, ros::Duration)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `YAML::Emitter::Write(std::string const&)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `svo::ImuHandler::loadCalibrationFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `YAML::LoadFile(std::string const&)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `YAML::detail::node_data::set_scalar(std::string const&)'
/home/odroid/svo_install_ws/install/lib/libvikit_cameras.so: undefined reference to `cv::imread(std::string const&, int)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::output_helper::publishQuadrocopterMarkers(ros::Publisher, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, int, int, double, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/home/odroid/svo_install_ws/install/lib/libvikit_common.so: undefined reference to `cv::Exception::Exception(int, std::string const&, std::string const&, std::string const&, int)'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::cameras::NCamera::loadFromYaml(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/odroid/svo_install_ws/install/lib/libsvo.so: undefined reference to `YAML::Emitter::PrepareIntegralStream(std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >&) const'
/home/odroid/svo_install_ws/install/lib/libvikit_cameras.so: undefined reference to `YAML::detail::node_data::empty_scalar'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
/home/odroid/svo_install_overlay_ws/devel/lib/libsvo_ros.so: undefined reference to `vk::output_helper::publishFrameMarker(ros::Publisher, Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time const&, int, int, double, ros::Duration)'
collect2: error: ld returned 1 exit status
CMakeFiles/svo_node.dir/build.make:653: recipe for target '/home/odroid/svo_install_overlay_ws/devel/lib/svo_ros/svo_node' failed
make[2]: *** [/home/odroid/svo_install_overlay_ws/devel/lib/svo_ros/svo_node] Error 1
CMakeFiles/Makefile2:284: recipe for target 'CMakeFiles/svo_node.dir/all' failed
make[1]: *** [CMakeFiles/svo_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target svo_nodelet
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
[svo_ros] <== '/home/odroid/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                                                                                                                                                                                                 
Failed   <== svo_ros [ 2 minutes and 0.1 seconds ]                                                                                                                                                                                                                              
[build] There were '1' errors:                                                                                                                                                                                                                                                  

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/odroid/svo_install_overlay_ws/build/svo_ros && /home/odroid/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/odroid/svo_install_overlay_ws/build/build_logs/svo_ros.log

Exited with return code: 2 
                                                                                                                                                                                                                                                                                
[build] Runtime: 2 minutes and 0.4 seconds 
```

I know the relevant libraries are installed on the system. I'm using the odroid XU4, the same processor that the demo was done on, with the armv71 architecture.

FPSychotic, 2018-02-03:
I have a quite simalar error, if not the same over imx6 with arm7a architecture. My knowledges are too basic in ros and linux

nightpool, 2018-02-07:
I fixed this by installing ubuntu 14.04 and ROS Indigo. See #25

closed 2018-02-07 by nightpool

MauUnterKarottenbaum, 2023-07-17:
check CMakeList