uzh-rpg/rpg_svo_example #22: OpenCV in ROS Kinetic compatibility problem

stanleywang-0-1, 2017-12-19:
Hi, recently I found SVO example can't be built due to some error as follows: (Ubuntu 16.04, ROS Kinetic)
``` 
Scanning dependencies of target svo_ros
[ 11%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_interface.cpp.o
[ 22%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_factory.cpp.o
[ 33%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_node_base.cpp.o
[ 44%] Building CXX object CMakeFiles/svo_ros.dir/src/visualizer.cpp.o
[ 55%] Linking CXX shared library /home/bcs/svo_install_overlay_ws/devel/lib/libsvo_ros.so
[ 55%] Built target svo_ros
Scanning dependencies of target svo_nodelet
Scanning dependencies of target svo_node
[ 77%] Building CXX object CMakeFiles/svo_node.dir/src/svo_node.cpp.o
[ 77%] Building CXX object CMakeFiles/svo_nodelet.dir/src/svo_nodelet.cpp.o
[ 88%] Linking CXX executable /home/bcs/svo_install_overlay_ws/devel/lib/svo_ros/svo_node
[100%] Linking CXX shared library /home/bcs/svo_install_overlay_ws/devel/lib/libsvo_nodelet.so
/usr/bin/ld: warning: libopencv_core3.so.3.2, needed by /home/bcs/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_highgui3.so.3.2, needed by /home/bcs/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_imgproc3.so.3.2, needed by /home/bcs/svo_install_ws/install/lib/libsvo.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_imgcodecs3.so.3.2, needed by /home/bcs/svo_install_ws/install/lib/libvikit_cameras.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_calib3d3.so.3.2, needed by /home/bcs/svo_install_ws/install/lib/libvikit_common.so, not found (try using -rpath or -rpath-link)
/home/bcs/svo_install_ws/install/lib/libvikit_common.socv::findHomography(cv::_InputArray const&, cv::_InputArray const&, int, double, cv::_OutputArray const&, int, double)
/home/bcs/svo_install_ws/install/lib/libsvo.socv::waitKey(int)
/home/bcs/svo_install_ws/install/lib/libsvo.socv::imshow(cv::String const&, cv::_InputArray const&)
collect2: error: ld returned 1 exit status
CMakeFiles/svo_node.dir/build.make:583: recipe for target '/home/bcs/svo_install_overlay_ws/devel/lib/svo_ros/svo_node' failed
make[2]: *** [/home/bcs/svo_install_overlay_ws/devel/lib/svo_ros/svo_node] Error 1
CMakeFiles/Makefile2:159: recipe for target 'CMakeFiles/svo_node.dir/all' failed
make[1]: *** [CMakeFiles/svo_node.dir/all] Error 2
make[1]: *** ....
[100%] Built target svo_nodelet
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
[svo_ros] <== '/home/bcs/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                 
Failed   <== svo_ros [ 1 minute and 37.4 seconds ]                              
[build] There were '1' errors:                                                  

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/bcs/svo_install_overlay_ws/build/svo_ros && /home/bcs/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/bcs/svo_install_overlay_ws/build/build_logs/svo_ros.log

Exited with return code: 2 
                                                                                
[build] Runtime: 1 minute and 38.0 seconds 
```
As suggested in a previous closed issue, I have checked my opencv libraries using ```ls -l /opt/ros/kinetic/lib/ | grep opencv```. And I got
```
lrwxrwxrwx 1 root root       23 11月  7 02:47 libopencv_aruco3.so -> libopencv_aruco3.so.3.3
lrwxrwxrwx 1 root root       25 11月  7 02:47 libopencv_aruco3.so.3.3 -> libopencv_aruco3.so.3.3.1
-rw-r--r-- 1 root root   308784 11月  7 02:47 libopencv_aruco3.so.3.3.1
lrwxrwxrwx 1 root root       24 11月  7 02:47 libopencv_bgsegm3.so -> libopencv_bgsegm3.so.3.3
lrwxrwxrwx 1 root root       26 11月  7 02:47 libopencv_bgsegm3.so.3.3 -> libopencv_bgsegm3.so.3.3.1
-rw-r--r-- 1 root root    76168 11月  7 02:47 libopencv_bgsegm3.so.3.3.1
lrwxrwxrwx 1 root root       29 11月  7 02:47 libopencv_bioinspired3.so -> libopencv_bioinspired3.so.3.3
lrwxrwxrwx 1 root root       31 11月  7 02:47 libopencv_bioinspired3.so.3.3 -> libopencv_bioinspired3.so.3.3.1
-rw-r--r-- 1 root root   277704 11月  7 02:47 libopencv_bioinspired3.so.3.3.1
lrwxrwxrwx 1 root root       25 11月  7 02:47 libopencv_calib3d3.so -> libopencv_calib3d3.so.3.3
lrwxrwxrwx 1 root root       27 11月  7 02:47 libopencv_calib3d3.so.3.3 -> libopencv_calib3d3.so.3.3.1
-rw-r--r-- 1 root root  1586904 11月  7 02:47 libopencv_calib3d3.so.3.3.1
lrwxrwxrwx 1 root root       24 11月  7 02:47 libopencv_ccalib3.so -> libopencv_ccalib3.so.3.3
lrwxrwxrwx 1 root root       26 11月  7 02:47 libopencv_ccalib3.so.3.3 -> libopencv_ccalib3.so.3.3.1
-rw-r--r-- 1 root root   434232 11月  7 02:47 libopencv_ccalib3.so.3.3.1
lrwxrwxrwx 1 root root       22 11月  7 02:47 libopencv_core3.so -> libopencv_core3.so.3.3
lrwxrwxrwx 1 root root       24 11月  7 02:47 libopencv_core3.so.3.3 -> libopencv_core3.so.3.3.1
-rw-r--r-- 1 root root 12978328 11月  7 02:47 libopencv_core3.so.3.3.1
lrwxrwxrwx 1 root root       21 11月  7 02:47 libopencv_cvv3.so -> libopencv_cvv3.so.3.3
[... 93 lines cut ...]
lrwxrwxrwx 1 root root       25 11月  7 02:47 libopencv_video3.so.3.3 -> libopencv_video3.so.3.3.1
-rw-r--r-- 1 root root  3609488 11月  7 02:47 libopencv_video3.so.3.3.1
lrwxrwxrwx 1 root root       25 11月  7 02:47 libopencv_videoio3.so -> libopencv_videoio3.so.3.3
lrwxrwxrwx 1 root root       27 11月  7 02:47 libopencv_videoio3.so.3.3 -> libopencv_videoio3.so.3.3.1
-rw-r--r-- 1 root root   163216 11月  7 02:47 libopencv_videoio3.so.3.3.1
lrwxrwxrwx 1 root root       27 11月  7 02:47 libopencv_videostab3.so -> libopencv_videostab3.so.3.3
lrwxrwxrwx 1 root root       29 11月  7 02:47 libopencv_videostab3.so.3.3 -> libopencv_videostab3.so.3.3.1
-rw-r--r-- 1 root root   347376 11月  7 02:47 libopencv_videostab3.so.3.3.1
lrwxrwxrwx 1 root root       21 11月  7 02:47 libopencv_viz3.so -> libopencv_viz3.so.3.3
lrwxrwxrwx 1 root root       23 11月  7 02:47 libopencv_viz3.so.3.3 -> libopencv_viz3.so.3.3.1
-rw-r--r-- 1 root root   394408 11月  7 02:47 libopencv_viz3.so.3.3.1
lrwxrwxrwx 1 root root       29 11月  7 02:47 libopencv_xfeatures2d3.so -> libopencv_xfeatures2d3.so.3.3
lrwxrwxrwx 1 root root       31 11月  7 02:47 libopencv_xfeatures2d3.so.3.3 -> libopencv_xfeatures2d3.so.3.3.1
-rw-r--r-- 1 root root  2879304 11月  7 02:47 libopencv_xfeatures2d3.so.3.3.1
lrwxrwxrwx 1 root root       26 11月  7 02:47 libopencv_ximgproc3.so -> libopencv_ximgproc3.so.3.3
lrwxrwxrwx 1 root root       28 11月  7 02:47 libopencv_ximgproc3.so.3.3 -> libopencv_ximgproc3.so.3.3.1
-rw-r--r-- 1 root root  1147448 11月  7 02:47 libopencv_ximgproc3.so.3.3.1
lrwxrwxrwx 1 root root       28 11月  7 02:47 libopencv_xobjdetect3.so -> libopencv_xobjdetect3.so.3.3
lrwxrwxrwx 1 root root       30 11月  7 02:47 libopencv_xobjdetect3.so.3.3 -> libopencv_xobjdetect3.so.3.3.1
-rw-r--r-- 1 root root   105160 11月  7 02:47 libopencv_xobjdetect3.so.3.3.1
lrwxrwxrwx 1 root root       24 11月  7 02:47 libopencv_xphoto3.so -> libopencv_xphoto3.so.3.3
lrwxrwxrwx 1 root root       26 11月  7 02:47 libopencv_xphoto3.so.3.3 -> libopencv_xphoto3.so.3.3.1
-rw-r--r-- 1 root root  3413520 11月  7 02:47 libopencv_xphoto3.so.3.3.1
```
It appeared that the OpenCV libraries in ROS Kinetic has updraded to 3.3.1 recently, and there is a compatibility problem with SVO binaries which tried to find 3.2 libraries.

zhangzichao, 2018-01-22:
Hi, sorry for the delay, I just updated the binaries, you can download the updated one with the same link. Let me know if it works for you.

FYI, there is also an issue with OpenCV 3.3.1:
https://github.com/ros-perception/vision_opencv/issues/193
and it is fixed in OpenCV 3.4. Therefore I would expect ROS will update to OpenCV 3.4 soon, which will break the binaries compatibility again.

amyking1102, 2018-01-23:
Thanks! It works!

BirBikram, 2018-04-03:
Hi, I am using 16.04, Kinetic  and  Opencv 3.4.1. And I have the same error as above, can this be fixed?
Thanks