uzh-rpg/rpg_svo_example #23: catkin build error

Lacdires, 2017-12-22:
This is my build log.

[[36msvo_ros[0m] [1m[34m==>[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make cmake_check_build_system[0m' [30m[1min[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros[0m'[0m
[[36msvo_ros[0m] [32m<==[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make cmake_check_build_system[0m' finished with return code '[1m0[0m'[0m
[[36msvo_ros[0m] [1m[34m==>[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j[0m' [30m[1min[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros[0m'[0m
[  0%] Built target topic_tools_generate_messages_py
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target topic_tools_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target nodelet_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target pcl_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target bond_generate_messages_py
[  0%] Built target nodelet_generate_messages_lisp
[  0%] Built target svo_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target nodelet_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target nodelet_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target visualization_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target topic_tools_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target bond_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target bond_generate_messages_eus
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target bond_generate_messages_lisp
[  0%] Built target nodelet_generate_messages_eus
[  0%] Built target topic_tools_generate_messages_cpp
[  0%] Built target topic_tools_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target pcl_msgs_generate_messages_eus
[  0%] Built target pcl_ros_gencfg
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target visualization_msgs_generate_messages_lisp
[  0%] Built target visualization_msgs_generate_messages_nodejs
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target svo_msgs_generate_messages_eus
[  0%] Built target visualization_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target tf_generate_messages_eus
[  0%] Built target tf_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target svo_msgs_generate_messages_nodejs
[  0%] Built target svo_msgs_generate_messages_cpp
[  0%] Built target svo_msgs_generate_messages_py
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target bond_generate_messages_nodejs
[  0%] Built target pcl_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target visualization_msgs_generate_messages_eus
[  0%] Built target pcl_msgs_generate_messages_py
[  0%] Built target pcl_msgs_generate_messages_cpp
make[2]: *** No rule to make target '/home/upboard/svo_install_ws/install/lib/libgflags.so', needed by '/home/upboard/svo_install_overlay_ws/devel/lib/libsvo_ros.so'.  Stop.
CMakeFiles/Makefile2:2050: recipe for target 'CMakeFiles/svo_ros.dir/all' failed
make[1]: *** [CMakeFiles/svo_ros.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
[[36msvo_ros[0m] [1m[31m<==[0m '[1m/home/upboard/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j[0m' [31mfailed with return code[0m '[1m2[0m'[0m

I use UPboard with Ubuntu 16.04.3 desktop-amd64 and install ros kinetic version. Did anyone has the same problem and be solved?

Romanenko-Serhii, 2018-01-24:
Yes, i had the same mistake.

How i solved that:
sudo apt-get install libgoogle-glog-dev  libgflags-dev

Then, you need find libgflags.so in /usr/lib/ this file (for me it was /usr/lib/x86_64-linux-gnu/libgflags.so), and make link to /home/upboard/svo_install_ws/install/lib/
ln -s /usr/lib/x86_64-linux-gnu/libgflags.so /home/upboard/svo_install_ws/install/lib/

Then you need to do same for libglog.so 

And again build you svo_ros.

If it will build, for start SVO2 i writing: 
LD_PRELOAD=/home/zichao/svo_install_ws/install/lib/libgflags.so.2.2.1 roslaunch svo_ros launch_name.launch

If don't use LD_PRELOAD, you will have mistake.

BirBikram, 2018-04-21:
I used your method and successfully built it in my machine.
Now I can't run the examples,as it shows error. I already tried using the way wrote:
LD_PRELOAD=/home/bikram/svo_install_ws/install/lib/libgflags.so.2.2.1 roslaunch svo_ros launch_name.launch

It just shows an error:
auto-starting new master
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
process[master]: started with pid [23622]
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 03e49b90-4526-11e8-9274-9cb6d0177e71
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
process[rosout-1]: started with pid [23635]
started core service [/rosout]
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
process[svo-2]: started with pid [23653]
/home/bikram/svo_install_overlay_ws/devel/lib/svo_ros/svo_node: error while loading shared libraries: libgflags.so.2.2: cannot open shared object file: No such file or directory
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
process[vis-3]: started with pid [23654]
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
process[svo_gui-4]: started with pid [23655]
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
[svo-2] process has died [pid 23653, exit code 127, cmd /home/bikram/svo_install_overlay_ws/devel/lib/svo_ros/svo_node __name:=svo __log:=/home/bikram/.ros/log/03e49b90-4526-11e8-9274-9cb6d0177e71/svo-2.log].
log file: /home/bikram/.ros/log/03e49b90-4526-11e8-9274-9cb6d0177e71/svo-2*.log
ERROR: ld.so: object 'svo_install_ws/install/lib/libgflags.so.2.2.1' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.

Any idea what is wrong :(

slinshady89, 2019-05-21:
Hey I deleted the two files because they're only text-files containing the lib's name:
```
file /home/localadmin/svo_install_ws/install/lib/libgflags.so
/home/localadmin/svo_install_ws/install/lib/libgflags.so: ASCII text, with no line terminators
```
If you compare them to another one you'll see the difference:
```
file /home/localadmin/svo_install_ws/install/lib/libvikit_ros.so
/home/localadmin/svo_install_ws/install/lib/libvikit_ros.so: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=3dfc6fd2293a152d5ad4a9dee6dd3f5bb38213e7, not stripped
```
so I deleted libglog.so and libgflag.so and did the symbolic links:

```ln -s /usr/lib/x86_64-linux-gnu/libglog.so /home/localadmin/svo_install_ws/install/lib/libglog.so```
```ln -s /usr/lib/x86_64-linux-gnu/libgflags.so /home/localadmin/svo_install_ws/install/lib/libgflags.so```

afterwards the ```catkin build``` was working for me.