uzh-rpg/rpg_svo_example #28: Monocular camera: scale changes (actually diminished) while moving around

bharat-robotics, 2018-02-04:
I am trying to move husky around the corridor with loops using monocular camera. But the map scale changes while moving the robot around. Are there any specific parameters I can tweak? I am not sure if adding IMU will help or not. Here are some images for reference...
[image]
[image]

skandaswaroop1998, 2019-12-11:
hi did you fix the issue? i am facing the same problem

kylelindgren, 2019-12-11:
Same ^^ any explanation or possible solutions would be much appreciated.

bharat-robotics, 2019-12-11:
> hi did you fix the issue? i am facing the same problem

Nope. This is one of the limitations of the package using the monocular camera. I am not sure it can be fixed easily.

kylelindgren, 2019-12-13:
That's unfortunate. But thank you for the response.