uzh-rpg/rpg_svo_example #29: ROS publish Odom

gftabor, 2018-02-07:
Currently to run a lot of ros built in systems /odom needs to be published in addition to TF. Could the live position and velocity estimate be published there as well? The velocity is the important part I am looking for.

Explanation can be seen here

> The navigation stack uses tf to determine the robot's location in the world and relate sensor data to a static map. However, tf does not provide any information about the velocity of the robot. Because of this, the navigation stack requires that any odometry source publish both a transform and a nav_msgs/Odometry message over ROS that contains velocity information. This tutorial explains the nav_msgs/Odometry message and provides example code for publishing both the message and transform over ROS and tf respectively.

http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom

zhangzichao, 2018-02-08:
Hi, if you need a specific message type, you can change the publisher in the visualizer, for example
https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/src/visualizer.cpp#L138

Publishing messages is independent of the binaries, so you should be able to adapt the code in this repository according to your own needs.

Regarding velocity, SVO2.0 is not a full VIO, therefore there is no velocity estimation maintained. The best we can do is probably to run a Kalman filter for velocity, similar to what is described in the paper:
http://rpg.ifi.uzh.ch/docs/JFR16_Faessler.pdf

gftabor, 2018-02-08:
Yup I understood the system was not trying to produce an accurate velocity but had assumed there had to be some internal representation of movement or position step since last frame that could be used to produce some velocity information. Would expect this velocity to be unfiltered and noisy.