uzh-rpg/rpg_svo_example #31: catkin build

hamididari, 2018-02-15:
Hi guys 
I try to build svo2 but i got this err:
`/root/svo_install_ws/install/lib/libsvo.so: file not recognized: File format not recognized
`
and this is my out put from build:

```
root@odroid-server:~/svo_install_overlay_ws# cat /root/svo_install_overlay_ws/build/build_logs/svo_ros.log
[svo_ros] ==> '/root/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/cmake /root/svo_install_overlay_ws/src/rpg_svo_example/svo_ros -DCATKIN_DEVEL_PREFIX=/root/svo_install_overlay_ws/devel -DCMAKE_INSTALL_PREFIX=/root/svo_install_overlay_ws/install -DCMAKE_BUILD_TYPE=Release' in '/root/svo_install_overlay_ws/build/svo_ros'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /root/svo_install_overlay_ws/devel
-- Using CMAKE_PREFIX_PATH: /root/svo_install_ws/install;/opt/ros/indigo
-- This workspace overlays: /root/svo_install_ws/install;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /root/svo_install_overlay_ws/build/svo_ros/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Using these message generators: gencpp;genlisp;genpy
-- Marking SHARED_LIBRARY "svo_ros" of package "svo_ros" for installation
-- Marking EXECUTABLE "svo_node" of package "svo_ros" for installation
-- Marking SHARED_LIBRARY "svo_nodelet" of package "svo_ros" for installation
-- Marking HEADER FILES in "include" folder of package "svo_ros" for installation
-- Marking SHARED CONTENT FOLDER "launch" of package "svo_ros" for installation
-- Marking SHARED CONTENT FOLDER "param" of package "svo_ros" for installation
-- Configuring done
-- Generating done
-- Build files have been written to: /root/svo_install_overlay_ws/build/svo_ros
[svo_ros] <== '/root/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/cmake /root/svo_install_overlay_ws/src/rpg_svo_example/svo_ros -DCATKIN_DEVEL_PREFIX=/root/svo_install_overlay_ws/devel -DCMAKE_INSTALL_PREFIX=/root/svo_install_overlay_ws/install -DCMAKE_BUILD_TYPE=Release' finished with return code '0'
[svo_ros] ==> '/root/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/root/svo_install_overlay_ws/build/svo_ros'
Scanning dependencies of target topic_tools_generate_messages_cpp
Scanning dependencies of target geometry_msgs_generate_messages_lisp
Scanning dependencies of target actionlib_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_py
Scanning dependencies of target actionlib_msgs_generate_messages_py
Scanning dependencies of target nodelet_generate_messages_cpp
Scanning dependencies of target nodelet_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_py
[  0%] [  0%] [  0%] Built target topic_tools_generate_messages_cpp
Built target actionlib_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_lisp
[  0%] [  0%] [  0%] [  0%] Built target actionlib_msgs_generate_messages_py
Built target nodelet_generate_messages_cpp
Built target geometry_msgs_generate_messages_py
[  0%] Built target nodelet_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_py
Scanning dependencies of target std_msgs_generate_messages_py
[... 77 lines cut ...]
Built target svo_msgs_generate_messages_lisp
[  0%] [  0%] Scanning dependencies of target topic_tools_generate_messages_py
Built target svo_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_py
Built target pcl_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] Built target dynamic_reconfigure_generate_messages_cpp
Built target pcl_msgs_generate_messages_lisp
Built target topic_tools_generate_messages_py
Scanning dependencies of target pcl_msgs_generate_messages_py
Scanning dependencies of target topic_tools_generate_messages_lisp
[  0%] Built target pcl_msgs_generate_messages_py
[  0%] Built target topic_tools_generate_messages_lisp
Scanning dependencies of target svo_ros
[ 16%] [ 33%] [ 50%] [ 66%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_interface.cpp.o
Building CXX object CMakeFiles/svo_ros.dir/src/svo_factory.cpp.o
Building CXX object CMakeFiles/svo_ros.dir/src/visualizer.cpp.o
Building CXX object CMakeFiles/svo_ros.dir/src/svo_node_base.cpp.o
Linking CXX shared library /root/svo_install_overlay_ws/devel/lib/libsvo_ros.so
/root/svo_install_ws/install/lib/libsvo.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
make[2]: *** [/root/svo_install_overlay_ws/devel/lib/libsvo_ros.so] Error 1
make[1]: *** [CMakeFiles/svo_ros.dir/all] Error 2
make: *** [all] Error 2
[svo_ros] <== '/root/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
```

closed 2018-03-04 by hamididari

dunfrey, 2018-05-17:
How this problem was fix?

hamididari, 2018-12-18:
@dunfrey it's because of wrong lib binary

Wingmore, 2020-08-26:
@hamididari What was the correct library? How did you fix it?