uzh-rpg/rpg_svo_example #32: Stereo fisheye cameras with minimal overlap not detecting features

petercheng00, 2018-03-01:
I'm testing a setup with stereo fisheye cameras that are facing in mostly opposite directions, such that there is minimal overlap between their views. It never successfully initializes though, and the number of features is always around 0-40, with most of them in the overlapping region, and they don't look like good features, despite plenty of corners elsewhere in the scene. If I run with either camera alone as a mono setup, number of features is often 180+ (max_fts), and results look good as expected.

Is there some aspect of initialization with 2 cameras that requires points in the overlapping region?
If I decrease the init_min_* parameters down to around 40, it will always crash right after "Init: Selected first frame."

gftabor, 2018-03-01:
So most of the time when people say stereo cameras they exclusively work on overlapped areas. The overlapped regions allow distance calculations to be made. 

**Traditionally**, each camera finds features in its image, Then the feature sets are compared to find overlapping features. Only overlapping features are then used for the rest of the calculations.

As someone trying to use stereo fisheye odometry you should check out
https://github.com/uzh-rpg/rpg_svo_example/issues/27

petercheng00, 2018-03-01:
In the paper there doesn't seem to be any such restriction, and the video also shows a 4-camera setup that doesn't appear to rely on stereo triangulation. Does the provided binary handle stereo cases specially, or perhaps I'm just misunderstanding somewhere?

I did see your issue, but I don't think I've encountered the same problem.