uzh-rpg/rpg_svo_example #40: TX2 Installation

BirBikram, 2018-04-21:
I am having trouble installing in TX2. Has anyone else tried doing it?
Here is the error I am getting:
Failed   <== svo_ros [ 2.4 seconds ]                                            
[build] There were '1' errors:                                                  

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/nvidia/svo_install_overlay_ws/build/svo_ros && /home/nvidia/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/cmake /home/nvidia/svo_install_overlay_ws/src/rpg_svo_example/svo_ros -DCATKIN_DEVEL_PREFIX=/home/nvidia/svo_install_overlay_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/svo_install_overlay_ws/install -DCMAKE_BUILD_TYPE=Release; cd -

# Path to log:
cat /home/nvidia/svo_install_overlay_ws/build/build_logs/svo_ros.log

Exited with return code: 1

gftabor, 2018-04-21:
https://github.com/uzh-rpg/rpg_svo_example/issues/8

BirBikram, 2018-04-21:
So any binaries for TX2?
Thanks
Bikram

gftabor, 2018-04-21:
So to the best of my knowledge they never released binaries that worked with TX2. For my system I ended up buying an ODROID XU4 to go with my TX2. The second machine isn't particularly expensive and over ethernet ROS makes it "just work".

BirBikram, 2018-04-21:
For my robot I was gonna use a TX2 to do the slam. If I use a XU4, it will be so much under powered. I guess I have to get TX1 and install MCPTAM.
:(