uzh-rpg/rpg_svo_example #44: catkin build failed

dunfrey, 2018-05-17:
Hi guys
I try to build svo2 in UDOO QUAD ubuntu 14.04 but i got this err:

[svo_ros] ==> '/home/udooer/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/home/udooer/svo_install_overlay_ws/build/svo_ros'
Scanning dependencies of target nodelet_generate_messages_lisp
Scanning dependencies of target sensor_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_cpp
Scanning dependencies of target svo_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] Built target sensor_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_cpp
Built target nodelet_generate_messages_lisp
[  0%] Built target svo_msgs_generate_messages_cpp
Scanning dependencies of target nodelet_generate_messages_py
Scanning dependencies of target tf_generate_messages_cpp
Scanning dependencies of target nav_msgs_generate_messages_lisp
[  0%] Scanning dependencies of target geometry_msgs_generate_messages_lisp
Built target nodelet_generate_messages_py
[  0%] [  0%] Built target tf_generate_messages_cpp
Built target nav_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
Scanning dependencies of target visualization_msgs_generate_messages_cpp
Scanning dependencies of target bond_generate_messages_py
Scanning dependencies of target bond_generate_messages_lisp
[  0%] Scanning dependencies of target std_msgs_generate_messages_lisp
Built target visualization_msgs_generate_messages_cpp
[  0%] [  0%] Built target bond_generate_messages_lisp
Scanning dependencies of target bond_generate_messages_cpp
Built target bond_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Scanning dependencies of target tf2_msgs_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_cpp
Built target bond_generate_messages_cpp
Scanning dependencies of target sensor_msgs_generate_messages_cpp
[  0%] Scanning dependencies of target actionlib_msgs_generate_messages_cpp
Built target tf2_msgs_generate_messages_lisp
[  0%] [  0%] Built target std_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_cpp
[  0%] Scanning dependencies of target nav_msgs_generate_messages_py
Built target actionlib_msgs_generate_messages_cpp
Scanning dependencies of target roscpp_generate_messages_lisp
Scanning dependencies of target nav_msgs_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_py
Scanning dependencies of target std_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Scanning dependencies of target tf_generate_messages_py
Built target roscpp_generate_messages_lisp
[  0%] Scanning dependencies of target nodelet_generate_messages_cpp
Built target std_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_py
Scanning dependencies of target roscpp_generate_messages_py
[  0%] Built target nodelet_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target geometry_msgs_generate_messages_py
[  0%] Scanning dependencies of target roscpp_generate_messages_cpp
Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_cpp
Scanning dependencies of target actionlib_msgs_generate_messages_py
Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
Scanning dependencies of target actionlib_msgs_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] [  0%] Built target actionlib_msgs_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_py
Scanning dependencies of target visualization_msgs_generate_messages_py
Scanning dependencies of target visualization_msgs_generate_messages_lisp
Scanning dependencies of target tf_generate_messages_lisp
[  0%] Built target visualization_msgs_generate_messages_lisp
[  0%] [  0%] Built target sensor_msgs_generate_messages_py
Built target visualization_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_cpp
Scanning dependencies of target tf2_msgs_generate_messages_cpp
Scanning dependencies of target svo_msgs_generate_messages_py
[  0%] [  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_cpp
Built target actionlib_generate_messages_cpp
[  0%] Built target svo_msgs_generate_messages_py
Scanning dependencies of target tf2_msgs_generate_messages_py
Scanning dependencies of target svo_msgs_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target svo_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_py
Scanning dependencies of target svo_ros
[ 16%] [ 33%] [ 50%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_interface.cpp.o
[ 66%] Building CXX object CMakeFiles/svo_ros.dir/src/svo_factory.cpp.o
Building CXX object CMakeFiles/svo_ros.dir/src/visualizer.cpp.o
Building CXX object CMakeFiles/svo_ros.dir/src/svo_node_base.cpp.o
In file included from /home/udooer/svo_install_ws/install/include/svo/direct/matcher.h:12:0,
                 from /home/udooer/svo_install_ws/install/include/svo/direct/depth_filter.h:17,
                 from /home/udooer/svo_install_ws/install/include/svo/svo.h:9,
                 from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/src/svo_factory.cpp:2:
/home/udooer/svo_install_ws/install/include/svo/common/types.h:6:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/svo_ros.dir/src/svo_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/include/svo_ros/svo_interface.h:11:0,
                 from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/include/svo_ros/svo_node_base.h:3,
                 from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/src/svo_node_base.cpp:1:
/home/udooer/svo_install_ws/install/include/svo/common/types.h:6:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
In file included from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/include/svo_ros/svo_interface.h:11:0,
                 from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/src/svo_interface.cpp:1:
/home/udooer/svo_install_ws/install/include/svo/common/types.h:6:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
In file included from /home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/src/visualizer.cpp:6:0:
/home/udooer/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/include/svo_ros/visualizer.h:23:33: fatal error: pcl_ros/point_cloud.h: No such file or directory
 #include <pcl_ros/point_cloud.h>
                                 ^
compilation terminated.
make[2]: *** [CMakeFiles/svo_ros.dir/src/svo_node_base.cpp.o] Error 1
make[2]: *** [CMakeFiles/svo_ros.dir/src/svo_interface.cpp.o] Error 1
make[2]: *** [CMakeFiles/svo_ros.dir/src/visualizer.cpp.o] Error 1
make[1]: *** [CMakeFiles/svo_ros.dir/all] Error 2
make: *** [all] Error 2
[svo_ros] <== '/home/udooer/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                        
Failed   <== svo_ros [ 1 minute and 26.4 seconds ]                                     
[build] There were '1' errors:                                                         

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/udooer/svo_install_overlay_ws/build/svo_ros && /home/udooer/svo_install_overlay_ws/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/udooer/svo_install_overlay_ws/build/build_logs/svo_ros.log

Exited with return code: 2

mhubii, 2018-09-05:
Hi dunfrey,

I have had a similar problem. It seems like the point cloud library is missing in your ros installation. Install it via:

`sudo apt-get install ros-kinetic-pcl-ros`

Then remove the svo_install_overlay_ws folder and follow the install instructions from scratch. This did the job for me. Hope it helps.

Martin

bigbellmercy, 2019-09-27:
In my case, after I installed 'ros-kinetic-pcl-ros', removed previous installation folders and then installed again from the beginning as the 'install.pdf' it succeeded.

Some more info for install:
1. When the './fix_path.sh' command is executed, there happened following three warnings. But it can be ignored as written in 'install.pdf'.
grep: install/include/opengv/test: No such file or directory
grep: install/include/opengv/test: No such file or directory
grep: install/include/opengv/test: No such file or directory

2. In the 'install.pdf' there is a mistype of 'cd' as below:
cd -r <extracted folder>/rpg_svo_example ~/svo_install_overlay_ws/src

The 'cd' should be changed 'cp' as below:
cp -r <extracted folder>/rpg_svo_example ~/svo_install_overlay_ws/src