uzh-rpg/rpg_svo_example #45: catkin build error

zxp771, 2018-05-31:
HI,I tried to install the SVO2.0 as the installation.pdf.
but when I excute the command:"catkin build" I would get this following problem.
I have modified the fix_path(remove the "zichao" to my own path)
In fact when I excute the ./fix_path it will shows"grep: install/include/opengv/test:nor this files or folder"

Profile:                     default
Extending:          [cached] /home/aero/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/aero/svo_install_overlay_ws
------------------------------------------------------------------------
Source Space:       [exists] /home/aero/svo_install_overlay_ws/src
Log Space:          [exists] /home/aero/svo_install_overlay_ws/logs
Build Space:        [exists] /home/aero/svo_install_overlay_ws/build
Devel Space:        [exists] /home/aero/svo_install_overlay_ws/devel
Install Space:      [unused] /home/aero/svo_install_overlay_ws/install
DESTDIR:            [unused] None
------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> svo_ros                                                          
_______________________________________________________________________________
Errors     << svo_ros:cmake /home/aero/svo_install_overlay_ws/logs/svo_ros/build.cmake.001.log
CMake Error at /home/aero/svo_install_overlay_ws/src/rpg_svo_example/svo_ros/CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin_simple.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "catkin_simple", but CMake did not find one.

  Could not find a package configuration file provided by "catkin_simple"
  with any of the following names:

    catkin_simpleConfig.cmake
    catkin_simple-config.cmake

  Add the installation prefix of "catkin_simple" to CMAKE_PREFIX_PATH or set
  "catkin_simple_DIR" to a directory containing one of the above files.  If
  "catkin_simple" provides a separate development package or SDK, be sure it
  has been installed.

cd /home/aero/svo_install_overlay_ws/build/svo_ros; catkin build --get-env svo_ros | catkin env -si  /usr/bin/cmake /home/aero/svo_install_overlay_ws/src/rpg_svo_example/svo_ros --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/aero/svo_install_overlay_ws/devel/.private/svo_ros -DCMAKE_INSTALL_PREFIX=/home/aero/svo_install_overlay_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
Failed     << svo_ros:cmake          [ Exited with code 1 ]                    
Failed    <<< svo_ros                [ 0.2 seconds ]                           
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.3 seconds total.

suraj-bijjahalli, 2019-08-13:
Hello, is there a solution to this problem. I am experiencing it as well.

MertEgeCAN, 2020-06-11:
In **\<your catkin workspace\>/src** folder:

`git clone https://github.com/catkin/catkin_simple.git`
`cd ..`
`catkin build`

Worked for me.

Wingmore, 2020-08-26:
try
`source svo_install_ws/install/setup.bash
`
If that doesnt work, also try 
`source /opt/ros/<your ros version>/setup.bash`