uzh-rpg/rpg_svo_example #47: Installation issue

immotri, 2018-07-20:
Hi,
I am trying to install SVO 2.0 in Ubuntu 16.04 and kinetic.
After following the installation guidelines I get stuck in "catkin build".
I get the following messages:

immo@aspire:~/svo_install_overlay_ws$ catkin build
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/immo
Source Space:       [exists] /home/immo/src
Build Space:        [exists] /home/immo/build
Devel Space:        [exists] /home/immo/devel
Install Space:     [missing] /home/immo/install
DESTDIR:                     None
-------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------- 
Found '1' packages in 0.0 seconds. 
Starting ==> svo_ros                                                                                                                  
                                                                                                                                      
[svo_ros] ==> '/home/immo/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/immo/build/svo_ros'
[ 88%] Built target actionlib_msgs_generate_messages_nodejs
[ 88%] Built target topic_tools_generate_messages_py
[ 88%] Built target std_srvs_generate_messages_py
[ 88%] Built target tf2_msgs_generate_messages_py
[ 88%] Built target std_msgs_generate_messages_nodejs
[ 88%] Built target geometry_msgs_generate_messages_cpp
[ 88%] Built target actionlib_msgs_generate_messages_lisp
[ 88%] Built target nav_msgs_generate_messages_nodejs
[ 88%] Built target roscpp_generate_messages_py
[ 88%] Built target nodelet_generate_messages_py
[ 88%] Built target roscpp_generate_messages_cpp
[ 88%] Built target tf2_msgs_generate_messages_nodejs
[ 88%] Built target nav_msgs_generate_messages_py
[ 88%] Built target std_msgs_generate_messages_py
[ 88%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 88%] Built target rosgraph_msgs_generate_messages_eus
[ 88%] Built target tf_generate_messages_lisp
[ 88%] Built target rosgraph_msgs_generate_messages_lisp
[ 88%] Built target pcl_msgs_generate_messages_nodejs
[ 88%] Built target std_srvs_generate_messages_nodejs
[ 88%] Built target geometry_msgs_generate_messages_py
[ 88%] Built target geometry_msgs_generate_messages_lisp
[ 88%] Built target bond_generate_messages_py
[ 88%] Built target std_srvs_generate_messages_lisp
[ 88%] Built target nodelet_generate_messages_lisp
[ 88%] Built target std_msgs_generate_messages_lisp
[ 88%] Built target tf_generate_messages_cpp
[ 88%] Built target actionlib_msgs_generate_messages_eus
[ 88%] Built target std_srvs_generate_messages_eus
[ 88%] Built target actionlib_generate_messages_lisp
[ 88%] Built target geometry_msgs_generate_messages_eus
[ 88%] Built target geometry_msgs_generate_messages_nodejs
[ 88%] Built target nodelet_generate_messages_cpp
[ 88%] Built target nav_msgs_generate_messages_lisp
[ 88%] Built target dynamic_reconfigure_generate_messages_lisp
[ 88%] Built target nodelet_generate_messages_nodejs
[... 101 lines cut ...]
CMakeFiles/svo_node.dir/build.make:539: recipe for target '/home/immo/devel/lib/svo_ros/svo_node' failed
make[2]: *** [/home/immo/devel/lib/svo_ros/svo_node] Error 1
CMakeFiles/Makefile2:155: recipe for target 'CMakeFiles/svo_node.dir/all' failed
make[1]: *** [CMakeFiles/svo_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
[svo_ros] <== '/home/immo/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
                                                                                                                                       
Failed   <== svo_ros [ 1.7 seconds ]                                                                                                  
[build] There were '1' errors:                                                                                                        

Failed to build package 'svo_ros' because the following command:

# Command to reproduce:
cd /home/immo/build/svo_ros && /home/immo/build/svo_ros/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/immo/build/build_logs/svo_ros.log

Exited with return code: 2 
                                                                                                                                      
[build] Runtime: 1.7 seconds 

-----------------------
Any advise would be highly appreciated!

gethubwy, 2019-07-10:
Hi immotri, did you solved it?

Ciao-li, 2021-08-28:
I ran into a similar problem, anyone have any good suggestions？😓

hipforth, 2022-04-09:
just 
`catkin clean --all`
`catkin build`