uzh-rpg/rpg_svo_example #51: SVO Monocular code does not work with images in "mono16" format (TUM-VI)

bharat-robotics, 2018-09-28:
The images from ([TUM-VI dataset](https://vision.in.tum.de/data/datasets/visual-inertial-dataset) cannot be loaded with error Unknown image type 2!. It would be better if you could add support for "mono16" images as well.

slinshady89, 2019-06-02:
I had the same issue.
go to [svo_interface](https://github.com/uzh-rpg/rpg_svo_example/blob/master/svo_ros/src/svo_interface.cpp) line 198 and change:

`image = cv_bridge::toCvCopy(msg)->image;`

to

` image = cv_bridge::toCvShare(imageMessage, sensor_msgs::image_encodings::BGR8)->image;`

But somehow the tracking of features is lost sometimes maybe due to my changes.

bharat-robotics, 2019-06-05:
I actually converted the dataset to mono8. Not sure about the SVO performance though.