uzh-rpg/rpg_svo_example #54: Installaton Issue

iremu, 2018-11-26:
```
grep: install/include/opengv/test: No such file or directory
sed: no input files
grep: install/include/opengv/test: No such file or directory
sed: no input files
grep: install/include/opengv/test: No such file or directory
sed: no input files
```

I get those errors when I run the fix_path.sh. How can I fix those?

RomeoStaub, 2019-05-08:
> Why are the instructions are made to create a new workspace, instead of working in an already existing one? Does anyone know how to make this run in an already existing workspace?

Were you able to install it in an existinag workspace?

slinshady89, 2019-05-16:
@RomeoStaub @fsblu  
I'm on that problem too. I think the overlaying workspace is necessary so you can use catkin in yours and caktin-tools in theirs. Building in the workspace described in the installation guide I get the issue that catkin_simple isn't found. 
that by the way is could be fixed by adding catkin_simple to the src folder in the workspace as a package. I'm quiet sure it's not the correct way to do