uzh-rpg/rpg_svo_example #60: Assertion error while running SVO in Gazebo

Jwenda, 2019-08-12:
Dear community,
I am facing the following error while running in Gazebo simulator. The SVO successfully initialize and runs for 5 seconds. Then it crashes on the following error. Does anyone know what could be the problem?

I have checked the timestamps of camera frames and they have equal time stamp. Do you know how to debug it when just binaries are available?

svo_node: /home/zichao/sources/svo_install_ws/src/rpg_svo_pro/svo/src/imu_handler.cpp:75: bool svo::ImuHandler::getMeasurements(double, double, bool, svo::ImuMeasurements&): Assertion `new_cam_timestamp > old_cam_timestamp' failed.
*** Aborted at 1565602624 (unix time) try "date -d @1565602624" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGABRT (@0x3e800006aea) received by PID 27370 (TID 0x7f2ec18eb700) from PID 27370; stack trace: ***
    @     0x7f2ef0b57f20 (unknown)
    @     0x7f2ef0b57e97 gsignal
    @     0x7f2ef0b59801 abort
    @     0x7f2ef0b4939a (unknown)
    @     0x7f2ef0b49412 __assert_fail
    @     0x7f2eeede4d07 svo::ImuHandler::getMeasurements()
    @     0x7f2ef14d97b4 svo::SvoInterface::setImuPrior()
    @     0x7f2ef14db10c svo::SvoInterface::stereoCallback()
    @     0x7f2ef14e8218 boost::detail::function::void_function_obj_invoker9<>::invoke()
    @     0x7f2ef14f44dd message_filters::CallbackHelper9T<>::call()
    @     0x7f2ef14efe44 message_filters::Signal9<>::call()
    @     0x7f2ef14f0463 message_filters::sync_policies::ExactTime<>::checkTuple()
    @     0x7f2ef14f2078 message_filters::Synchronizer<>::cb<>()
    @     0x7f2ef14ef8b2 image_transport::SubscriberFilter::cb()
    @     0x7f2ec29a51eb image_transport::RawSubscriber::internalCallback()
    @     0x7f2eef05905e boost::detail::function::void_function_obj_invoker1<>::invoke()
    @     0x7f2ec29a94c1 ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f2ef1871792 ros::SubscriptionQueue::call()
    @     0x7f2ef181be8c ros::CallbackQueue::callOneCB()
    @     0x7f2ef181d27b ros::CallbackQueue::callAvailable()
    @     0x7f2ef14dcd0c svo::SvoInterface::stereoLoop()
    @     0x7f2ef11df66f (unknown)
    @     0x7f2eef84f6db start_thread
    @     0x7f2ef0c3a88f clone
    @                0x0 (unknown)
[svo-1] process has died [pid 27370, exit code -6, cmd /home/mrs/svo_install_overlay_ws/devel/lib/svo_ros/svo_node __name:=svo __log:=/home/mrs/.ros/log/9f858c5e-b9a5-11e9-bfb1-1c697a0252d2/svo-1.log].
log file: /home/mrs/.ros/log/9f858c5e-b9a5-11e9-bfb1-1c697a0252d2/svo-1*.log

Thanks,
Jan

Jwenda, 2019-09-10:
I have solved the problem. It was caused by Gazebo due to repetitive publishing of the same image, with the same timestamp, did not know about that. Now just filter them out.

closed 2019-09-10 by Jwenda

atharva417, 2020-08-02:
I'm getting the same error, can u plz describe in bit detail, what exactly u did?

bednarhonza, 2020-08-06:
Hello @atharva417,
I have created a simple message throw node, that just check the timestamp of the message. If the timestamp of the new message is not higher than the old message timestamp, throw it away.