uzh-rpg/rpg_svo_example #72: Build for Ubuntu 18 + ROS Melodic

thomas-kuruvila, 2021-01-04:
I have been successful  in setting up the estimator in ubuntu 16 with ROS Kinetic. However application requires dependencies on Ubuntu 18. 
I am trying to set it up for ubuntu 18, but facing serious dependency error. @zhangzichao  Can this be successfully done?

thomas-kuruvila, 2021-01-06:
I was able to successfully build svo_ros for ubuntu 18, though its not an easy task.

closed 2021-01-06 by thomas-kuruvila

Abrar3616, 2021-01-27:
Hi Thomas.
Hope you are well. I would also like to know how you built it for melodic with ubuntu 18 if you could please assist in this i would be very grateful. Many Thanks

Regards

thomas-kuruvila, 2021-01-28:
The issues i faced were all version mismatch or absence of certain libraries. Even though i cannot point a single good happy path, clonning and building the repo, and solving for the build issues worked out for me.

Abrar3616, 2021-01-28:
Alright thanks i got everything else done but i got this error and i don't know how to deal with it. Could you please help me in this

[image]

Abrar3616, 2021-01-28:
ok so i fixed this issue by removing the -Werror flag in cmake file. now i have the following error: 

[image]

thomas-kuruvila, 2021-01-28:
I remember encountering the same issue, even when i had opencv3 installed. I observed that the shared object files i had were not libopencv_core3.so.3.3. Instead my system had libopencv_core.so.3.3. The path i took may not be the most ideal. I checked up with my previous machine, which was running ubuntu16.04, which had the entire shared files. I copied these to /usr/lib. 
Hope this works for you!