uzh-rpg/rpg_svo_example #77: arm32 build fails

jbourne15, 2021-06-18:
I get the following build error: 

`*** Error in `/usr/bin/c++': double free or corruption (top): 0x0009dfa0 ***`

I successfully built on ubuntu 16.04.  From what I see in the instructions to build are very similar (differ on setting `export ARM_ARCHITECTURE=armv7-a`)

Any ideas of what is wrong?

jbourne15, 2021-06-18:
Maybe because my system is `armv7l`?

escobarhector, 2021-06-21:
I got it to build by upgrading g++ and gcc with the following 

`apt-get install software-properties-common -y
add-apt-repository ppa:ubuntu-toolchain-r/test -y && apt-get update \
  && apt-get install gcc-4.9 -y && apt-get install g++-4.9 -y \
  && cd /usr/bin && ln -s /usr/bin/g++-4.9 /usr/bin/g++ -f \
  && ln -s /usr/bin/gcc-4.9 /usr/bin/gcc -f`

closed 2021-06-21 by jbourne15